meshtastic/python #644: Meshtastic Python 2.3.14 fails on devices with 2.4.x firmware

resiliencetheatre, 2024-08-04:
Several errors on python 2.3.14 based script after upgraded to 2.4.x firmware. Firmware version 2.3.13 is working. Verified with Heltec V3 and tlora-v2-1-1-6-tcxo hardware.

```
ERROR:root:Error while handling message from radio 'utf-8' codec can't decode byte 0x94 in position 79: 'utf-8' codec can't decode byte 0x94 in position 79: invalid start byte in field: meshtastic.LogRecord.message
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x94 in position 79: 'utf-8' codec can't decode byte 0x94 in position 79: invalid start byte in field: meshtastic.LogRecord.message
```

```
ERROR:root:Error while handling message from radio Truncated message.
google.protobuf.message.DecodeError: Truncated message.
```

```
ERROR:root:Error while handling message from radio Wrong wire type in tag.
google.protobuf.message.DecodeError: Wrong wire type in tag.
```

ianmcorvidae, 2024-08-09:
Hi, sorry for the delay in looking at this. I'm unsure if this is python, firmware, or both, so it'd be good to get more information (particularly the exact protobuf bytes it's trying to handle when the failure happens). If you're able to reproduce this with something using the regular CLI, with `--debug` and `--seriallog` on, and provide the whole log, it might be easier to figure out what's going on and where. If not, I can probably advise on how to add some debug logging to your custom script, to similar effect.

resiliencetheatre, 2024-08-10:
It seems that this error is gone after I upgraded radio to Meshtastic Firmware 2.4.2.5b45303 Alpha.

closed 2024-08-10 by resiliencetheatre