meshtastic/python #645: Setting single location parameter clears other location params

ChrisRomp, 2024-08-04:
Perhaps this is a documentation bug if the behavior is intended, but if you pass a location parameter such as `--setlat` and do not include longitude or altitude, the unspecified values are cleared. In order to set a position via the CLI, it appears to be necessary to specify `--setlat`, `--setlon`, and `--setalt` args together.

Option 1: The CLI should only set/unset values that are specified.

Option 2: Update the [CLI documentation](https://meshtastic.org/docs/software/python/cli/#--setlat-setlat) to specify that all three parameters should be set together.

ianmcorvidae, 2024-08-04:
Yeah, probably the documentation should be updated to specify that if any of lat/lon/alt aren't included they'll be set to 0. There's not really a very clean way to only update one value at a time.

ianmcorvidae, 2024-12-21:
Lost track of this, but the documentation for this was updated in 6be39695778473039dc1cd36503ae2653df817a9 which I think is what needs doing here. I do also need to push the changes to the main docs site, but that's separate.

closed 2024-12-21 by ianmcorvidae