meshtastic/python #649: [Bug]: Remote admin --seturl functionality not changing LoRa freq slot.

ayysasha, 2024-08-18:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.4.0.46d7b82

### Description

When attempting to set a new url on a remote note, which includes a change in LoRa channel slot, the remote node fails to take the new Channel Slot settings and only applies the new channel and PSK settings.

This is using --dest xxx and --seturl xxxx

The new URL contains both a new primary and new LoRa freq slot, but  the same psk and slot for the admin channel.

Reproduction Steps:

meshtastic --dest xxx --seturl https://meshtastic.org/e/#CjISIDhLvMAdwCLRgb82uGEh4fuWty5Vv3Qifp1q-0jWTLGhGgpCdXJudGFzdGljOgIIIAopEiCs-O6_F1BdAZ80_FJPUw2dVGwyD-HNujCNeI5TGVJmsRoFYWRtaW4SEwgBEAY4AUADSAFQHlgPaAHABgE

(The node is orig. on channel 39 and this URL intends to switch it to channel 15, shortfast, with a few channels)

Behaviour: The channels and but not the lora frequency slot.

I have captured one of these failures with meshtastic --noproto in the log output below, which is from the target device.

### Relevant log output

```Shell
DEBUG | 08:55:11 73 [Router] Module 'Admin' wantsPacket=1
INFO  | 08:55:11 73 [Router] Received Admin from=0x49bdab6e, id=0x5446db92, portnum=6, payloadlen=2
INFO  | 08:55:11 73 [Router] Client is getting channel 0
INFO  | 08:55:11 73 [Router] Asked module 'Admin' to send a response
DEBUG | 08:55:11 73 [Router] Module 'routing' wantsPacket=1
INFO  | 08:55:11 73 [Router] Received routing from=0x49bdab6e, id=0x5446db92, portnum=6, payloadlen=2
DEBUG | 08:55:11 73 [Router] Routing sniffing (id=0x5446db92 fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x1 Portnum=6 WANTRESP rxtime=1723971311 rxSNR=11 rxRSSI=-30 hopStart=3)
DEBUG | 08:55:11 73 [Router] Delivering rx packet (id=0x5446db92 fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x1 Portnum=6 WANTRESP rxtime=1723971311 rxSNR=11 rxRSSI=-30 hopStart=3)
DEBUG | 08:55:11 73 [Router] Update DB node 0x49bdab6e, rx_time=1723971311
DEBUG | 08:55:11 73 [Router] Forwarding to phone (id=0x5446db92 fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x1 Portnum=6 WANTRESP rxtime=1723971311 rxSNR=11 rxRSSI=-30 hopStart=3)
DEBUG | 08:55:11 73 [Router] Module 'routing' considered
DEBUG | 08:55:11 73 [Router] Sending response (id=0x1cfec3f6 fr=0x7c to=0x6e, WantAck=0, HopLim=2 Ch=0x1 Portnum=6 requestId=5446db92 rxtime=1723971311 priority=70)
DEBUG | 08:55:11 73 [Router] Update DB node 0x7c546c7c, rx_time=1723971311
DEBUG | 08:55:11 73 [Router] Add packet record (id=0x1cfec3f6 fr=0x7c to=0x6e, WantAck=0, HopLim=2 Ch=0x1 Portnum=6 requestId=5446db92 rxtime=1723971311 priority=70)
DEBUG | 08:55:11 73 [Router] Using AES256 key!
DEBUG | 08:55:11 73 [Router] Software encrypt fr=7c546c7c, num=1cfec3f6, numBytes=63!
DEBUG | 08:55:11 73 [Router] enqueuing for send (id=0x1cfec3f6 fr=0x7c to=0x6e, WantAck=0, HopLim=2 Ch=0x56 encrypted rxtime=1723971311 hopStart=2 priority=70)
DEBUG | 08:55:11 73 [Router] txGood=2,rxGood=5,rxBad=0
DEBUG | 08:55:11 73 [RadioIf] Starting low level send (id=0x1cfec3f6 fr=0x7c to=0x6e, WantAck=0, HopLim=2 Ch=0x56 encrypted rxtime=1723971311 hopStart=2 priority=70)
DEBUG | 08:55:11 73 [RadioIf] (bw=250, sf=9, cr=4/5) packet symLen=2 ms, payloadSize=79, time 240 ms
DEBUG | 08:55:11 73 [RadioIf] AirTime - Packet transmitted : 240ms
DEBUG | 08:55:12 74 [RadioIf] Completed sending (id=0x1cfec3f6 fr=0x7c to=0x6e, WantAck=0, HopLim=2 Ch=0x56 encrypted rxtime=1723971311 hopStart=2 priority=70)
DEBUG | 08:55:12 74 [RadioIf] Starting low level send (id=0x5446db91 fr=0x6e to=0xff, WantAck=0, HopLim=2 Ch=0x9a encrypted rxSNR=11 rxRSSI=-30 hopStart=3 priority=64)
DEBUG | 08:55:12 74 [RadioIf] (bw=250, sf=9, cr=4/5) packet symLen=2 ms, payloadSize=37, time 145 ms
DEBUG | 08:55:12 74 [RadioIf] AirTime - Packet transmitted : 145ms
DEBUG | 08:55:12 74 [RadioIf] Completed sending (id=0x5446db91 fr=0x6e to=0xff, WantAck=0, HopLim=2 Ch=0x9a encrypted rxSNR=11 rxRSSI=-30 hopStart=3 priority=64)
DEBUG | 08:55:12 74 [RadioIf] (bw=250, sf=9, cr=4/5) packet symLen=2 ms, payloadSize=24, time 115 ms
DEBUG | 08:55:12 74 [RadioIf] Lora RX (id=0x5446db93 fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x56 encrypted rxSNR=10.25 rxRSSI=-31 hopStart=3)
DEBUG | 08:55:12 74 [RadioIf] AirTime - Packet received : 115ms
DEBUG | 08:55:12 74 [Router] Add packet record (id=0x5446db93 fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x56 encrypted rxSNR=10.25 rxRSSI=-31 hopStart=3)
DEBUG | 08:55:12 74 [Router] Using channel 1 (hash 0x56)
DEBUG | 08:55:12 74 [Router] Using AES256 key!
DEBUG | 08:55:12 74 [Router] Software encrypt fr=49bdab6e, num=5446db93, numBytes=8!
DEBUG | 08:55:12 74 [Router] decoded message (id=0x5446db93 fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x1 Portnum=6 WANTRESP rxtime=1723971312 rxSNR=10.25 rxRSSI=-31 hopStart=3)
DEBUG | 08:55:12 74 [Router] handleReceived(REMOTE) (id=0x5446db93 fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x1 Portnum=6 WANTRESP rxtime=1723971312 rxSNR=10.25 rxRSSI=-31 hopStart=3)
DEBUG | 08:55:12 74 [Router] Module 'Admin' wantsPacket=1
INFO  | 08:55:12 74 [Router] Received Admin from=0x49bdab6e, id=0x5446db93, portnum=6, payloadlen=2
[... 287 lines cut ...]
DEBUG | 08:55:22 84 [Router] txGood=12,rxGood=14,rxBad=0
DEBUG | 08:55:22 84 [RadioIf] (bw=250, sf=9, cr=4/5) packet symLen=2 ms, payloadSize=46, time 166 ms
DEBUG | 08:55:22 84 [RadioIf] Lora RX (id=0x5446db9c fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x56 encrypted rxSNR=10.75 rxRSSI=-31 hopStart=3)
DEBUG | 08:55:22 84 [RadioIf] AirTime - Packet received : 166ms
DEBUG | 08:55:22 84 [Router] Add packet record (id=0x5446db9c fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x56 encrypted rxSNR=10.75 rxRSSI=-31 hopStart=3)
WARN  | 08:55:22 84 [Router] No suitable channel found for decoding, hash was 0x56!
DEBUG | 08:55:22 84 [Router] packet decoding failed or skipped (no PSK?) (id=0x5446db9c fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x56 encrypted rxtime=1723971322 rxSNR=10.75 rxRSSI=-31 hop
DEBUG | 08:55:22 84 [Router] Module 'routing' wantsPacket=1
INFO  | 08:55:22 84 [Router] Received routing from=0x49bdab6e, id=0x5446db9c, portnum=30, payloadlen=26205
DEBUG | 08:55:22 84 [Router] Routing sniffing (id=0x5446db9c fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x56 encrypted rxtime=1723971322 rxSNR=10.75 rxRSSI=-31 hopStart=3)
DEBUG | 08:55:22 84 [Router] Delivering rx packet (id=0x5446db9c fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x56 encrypted rxtime=1723971322 rxSNR=10.75 rxRSSI=-31 hopStart=3)
DEBUG | 08:55:22 84 [Router] Forwarding to phone (id=0x5446db9c fr=0x6e to=0x7c, WantAck=0, HopLim=3 Ch=0x56 encrypted rxtime=1723971322 rxSNR=10.75 rxRSSI=-31 hopStart=3)
WARN  | 08:55:22 84 [Router] No suitable channel found for decoding, hash was 0x56!
DEBUG | 08:55:22 84 [Router] Module 'routing' considered
DEBUG | 08:55:22 84 [RadioIf] Starting low level send (id=0x1cfec3ff fr=0x7c to=0x6e, WantAck=0, HopLim=2 Ch=0xef encrypted rxtime=1723971322 hopStart=2 priority=120)
DEBUG | 08:55:22 84 [RadioIf] (bw=250, sf=9, cr=4/5) packet symLen=2 ms, payloadSize=27, time 123 ms
DEBUG | 08:55:22 84 [RadioIf] AirTime - Packet transmitted : 123ms
DEBUG | 08:55:22 84 [RadioIf] Completed sending (id=0x1cfec3ff fr=0x7c to=0x6e, WantAck=0, HopLim=2 Ch=0xef encrypted rxtime=1723971322 hopStart=2 priority=120)
DEBUG | 08:55:23 85 [Power] Battery: usbPower=0, isCharging=0, batMv=4099, batPct=93
DEBUG | 08:55:43 105 [Power] Battery: usbPower=0, isCharging=0, batMv=4100, batPct=93
DEBUG | 08:56:03 125 [Power] Battery: usbPower=0, isCharging=0, batMv=4101, batPct=93
DEBUG | 08:56:23 145 [Power] Battery: usbPower=0, isCharging=0, batMv=4102, batPct=93
DEBUG | 08:56:43 165 [Power] Battery: usbPower=0, isCharging=0, batMv=4105, batPct=93
DEBUG | 08:57:03 185 [Power] Battery: usbPower=0, isCharging=0, batMv=4107, batPct=94
```

ayysasha, 2024-08-19:
Also FYI for the CLI its 2.3.15

jp-bennett, 2024-08-22:
My guess is that the channel setting gets applied first, then the LoRa command is missed because the channel has already changed.