meshtastic/python #651: 2.5 updates

jp-bennett, 2024-08-21:
WIP List of work needed for 2.5:
- [ ] import 2.5 protobufs
- [ ] expose security config
- [ ] show public keys for known nodes
- [ ] add support for remote admin session key (probably by making a request for get_sessionkey config before sending any set_x remote admin commands)
- [ ] add support for remote admin over PKI

ianmcorvidae, 2024-08-22:
WIP branch ( https://github.com/meshtastic/python/tree/2.5 ) has the current protobufs `master` imported and regenerated, and security config should, hopefully, be accessible as of https://github.com/meshtastic/python/commit/58466f2ab77ab783f3976202a2505d398784c9d7

closed 2024-09-11 by ianmcorvidae

ianmcorvidae, 2024-09-11:
I didn't get around to doing `--set-time`, so that'll have to be 2.5.1 or a postrelease version. Which is fine.