meshtastic/python #662: Changing channel preset over admin channel fails on congested networks

jonootto, 2024-08-30:
> meshtastic --host 192.168.1.10 --dest '!xxx' --ch-shortfast
Connected to radio
Requesting channel 0 info from remote node (this could take a while)
Requesting channel 1 info from remote node (this could take a while)
Requesting channel 2 info from remote node (this could take a while)

Attempting to set the channel preset over admin channel goes into a loop of requesting channel information which almost inevitably times out in a congested mesh, is this necessary? Could an option be added to skip fetching current config?

jonootto, 2024-08-30:
This was after about 20 attempts where the majority hung on channel 0, none successfully got the point of sending the new config

ianmcorvidae, 2024-08-31:
I think those `--ch-x` commands should be able to work over admin too, but I don't know why they're requesting every channel. I'll need to check if they're doing something other than just setting the preset but hopefully we can disable the channel fetching at least. Until then, using `--set` is a good idea, since it'll only need to fetch the config for that.

jonootto, 2024-08-31:
Thanks folks, confirm the --set option has worked fine, do you want to keep this open to look into the --ch commands?

ianmcorvidae, 2024-08-31:
Yeah, I'd like to check on those still. If they're really just setting the preset, maybe we can streamline some stuff. If they're doing more, we should document the differences, so either way let's leave this open, and thanks for the report!

PR #668 (2024-09-12): Change modem preset shortcuts to not request channels, and to request remote config when needed
Fixes #662

This should change the `--ch-<preset>` commands to behave equivalently to `--set lora.modem_preset <preset>`
closed 2024-09-12 by ianmcorvidae