meshtastic/python #669: Heltec V3 Crash with Python CLI

BenParker2022, 2024-09-08:
I have narrowed it down to a problem with the API. i have tried different computers, different installations of drivers, meshtastic, python, pip, and also changed between two devices. when trying to post a command like "meshtastic --info" the device shuts down and starts back up again. when using the command "meshtastic --noproto" the device stays on and displays information. I am new at this and if I messed something up or did something wrong let me know. but have not seen this problem on any forum or discord group.

pdxlocations, 2024-09-08:
Have you tried different USB cables?  USB ports?   I'm thinking it could be a power issue.  Have you updated your CLI?  FW?  Tried a full erase and reinstall of the firmware on the node? I'm running a V3 without issues using the CLI.

BenParker2022, 2024-09-08:
I have tried different usb ports. The usb cable is an interesting thought though. I only have one that allows me to connect to the node so I will go buy a new one. I have tried the update and the erase and flash for the latest 3 stable firmware versions. Not sure if this is anything but When running the noproto I saw a text in red that said “could not open / read /oem/oem.proto. Really got thrown in a loop here. Trying to learn all this stuff at the same time is fun but at some points like right now is aggravating.

BenParker2022, 2024-09-08:
Also I tried turning on my battery to try and see if the problem was the power supply. It did not change it turning off when sending a command.

thebentern, 2024-09-08:
Are you running on windows? I've definitely noticed some inconsistently bad handling of DTR/RTS by pySerial in Windows vs OSX / Linux.

BenParker2022, 2024-09-08:
Yes Windows 11. I'm not familiar with this, are you saying the computer's side is the problem? Also what driver version is everyone running for Silicon labs?    Should I try using Linux to see if windows is the culprit?

slash128v6, 2024-09-10:
I am having this same issue. I have 2x Heltec V3 devices that flash fine but when I attempt to use the CLI the devices just restart. Not showing "rebooting" but the screen goes blank and the device restarts. No settings are saved. Tried several cables (they all work for flashing). Tried several laptops all running Win11 and Python 3.12.

BenParker2022, 2024-09-10:
Do you also have any VPN or firewall configuration? Trying to determine if that is the issue.

slash128v6, 2024-09-10:
No VPN. I tried with Windows Defender firewall enabled and disabled, same result. Seems to be a Windows thing though, as CLI works for me using Linux.

BenParker2022, 2024-09-10:
I am able to run meshtastic --noproto fine. I think it is a problem with the API implementation possibly.

slash128v6, 2024-09-10:
Same for me.

slash128v6, 2024-09-10:
Yes, CLI using Ubuntu Linux 22.04 in VMware Workstation 17 on Windows 11 works for me.

BenParker2022, 2024-09-20:
I was able to run it fine on a computer running windows 10. I think it might be a problem with windows thinking the node is a HID mouse or something. Definitely not the actual python script or the API. Still trying to figure this out.

ianmcorvidae, 2024-12-21:
Hi; I'm not sure if you've made any progress on debugging this, but I'm trying to clean out issues a bit. I'll close this for now, along with #670 that seems to be the same issue. But if you can figure anything out, definitely let me know or reopen.