meshtastic/python #670: Problems with CLI with Heltec v3 board

BenParker2022, 2024-09-08:
when trying a command like meshtastic --info or meshtastic --ack the device shuts down. here are my stats that i got from the CLI. the only thing that works is -noproto. I have the proper drives and updated meshtastic, python, pip, and pytap. 

 System: Windows
   Platform: Windows-10-10.0.22631-SP0
   Release: 10
   Machine: AMD64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v2.4.1
 Executable: C:\Users\******\AppData\Local\Programs\Python\Python311\Scripts\meshtastic
 Python: 3.11.7 CPython MSC v.1937 64 bit (AMD64)

Please add the output from the command: meshtastic --info

C:\Users\bendp>meshtastic --info

C:\Users\bendp>meshtastic --listen
DEBUG file:__main__.py common line:1131 Not logging serial output
DEBUG file:serial_interface.py __init__ line:35 ports:['COM4']
DEBUG file:serial_interface.py __init__ line:46 Connecting to COM4
DEBUG file:stream_interface.py __reader line:147 in __reader()
DEBUG file:stream_interface.py _sendToRadioImpl line:110 Sending: want_config_id: 803497996
DEBUG file:stream_interface.py _sendToRadioImpl line:115 sending header:b'\x94\xc3\x00\x06' b:b'\x18\x8c\xd0\x91\xff\x02'
DEBUG file:mesh_interface.py _handleFromRadio line:963 in mesh_interface.py _handleFromRadio() fromRadioBytes: b'@\x01'
DEBUG file:mesh_interface.py _handleFromRadio line:967 Received from radio: id: 3804064768
rebooted: true      # not sure if this is the problem but it crashes about 7 lines down

DEBUG file:stream_interface.py _sendToRadioImpl line:110 Sending: want_config_id: 3897468303
DEBUG file:stream_interface.py _sendToRadioImpl line:115 sending header:b'\x94\xc3\x00\x06' b:b'\x18\x8f\xcb\xba\xc2\x0e'
DEBUG file:mesh_interface.py _handleFromRadio line:963 in mesh_interface.py _handleFromRadio() fromRadioBytes: b'27\n$Client wants config, nonce=803497996\x1a\rSerialConsole \x14'

It crashes right after this. I don't think it is a power issue because when i connect a battery it still does it, thought i do not know if it only charges the battery when plugged in via serial. please help this is super annoying as I am trying to get this setup for remote node use. not sure if it is a programing bug in python or a issue on windows side. I am going to get a better USB C cable to see if that changes it. tried to connect on two separate computers with the same results

kf4mot, 2024-09-13:
Same for me.
I know I have other issues, at least a serial permission Windows problem since I can't get CLI to really work at all. I don't see other modules rebooting like this Heltec one though.

BenParker2022, 2024-09-17:
It does not have reboot on the screen. Light flashes and turns to black for a couple seconds. Then comes back on with the meshtastic logo. The most recent stable version 2.4.2.5.

BenParker2022, 2024-09-17:
I have two board and they both do the same thing. Heltec v3

BenParker2022, 2024-09-20:
I was able to run it fine on a computer running windows 10. I think it might be a problem with windows thinking the node is a HID mouse or something. Definitely not the actual python script or the API. Still trying to figure this out.

closed 2024-12-21 by ianmcorvidae