meshtastic/python #682: Python CLI does not seem to retrieve stored messages on connection

kop316, 2024-10-11:
Hello!

I have been building a [GUI](https://gitlab.com/kop316/gtk-meshtastic-client) on top of your library, and I have been trying to figure out how to retrieve any messages that the device received and has stored while it is not connected to the python CLI. Most of the examples I have seen are some form of:

```
import meshtastic.serial_interface
from pubsub import pub
import json
import time

def onReceive(packet, interface):
    packet = json.dumps(packet, indent=2, default=lambda s: " ".join(str(s).split()))
    print(f"{packet} \n\n")

pub.subscribe(onReceive, 'meshtastic.receive')
interface = meshtastic.serial_interface.SerialInterface()

while True:
    time.sleep(1)
```

Such as from: https://github.com/pdxlocations/Meshtastic-Python-Examples . However, from what I can tell, `meshtastic.receive` only sends messages that are handled while there is an active connection, and does not check for any stored messages. 

In looking through the notable properties: https://github.com/meshtastic/python/blob/master/meshtastic/__init__.py#L8 , I likewise do not see anything similar to `getNodes()` (that has all of the stored nodes) that would be for stored messages.

In testing the device with the Android App, it seems to be able to pull any stored messages from the device on first connection.

So I am wondering, is there a way to retrieve stored messages, or is this missing functionality?

ianmcorvidae, 2024-10-11:
The only stored messages that exist to be fetched from the device are really just a queue -- up to about 30 packets worth -- that's kept/maintained when no client is connected. They're sent right after connecting, just like any further packets that get received after that; there's no way of fetching stored messages since there really aren't any, just messages waiting in a queue for the next client.

So a pubsub callback is indeed the way to get the limited number that might be available. Data persistence is done on the client side with Meshtastic, not on the device itself, so that'd be getting added to your GUI code. The Android and other apps are doing the same thing, getting sent queued messages by the device just after connecting.

kop316, 2024-10-11:
Hello!

I appreciate the confirmation on that. I think I may see the issue. When I run the below code:

```
import meshtastic.serial_interface
from pubsub import pub
import json
import time

def onReceive(packet, interface):
    packet = json.dumps(packet, indent=2, default=lambda s: " ".join(str(s).split()))
    print(f"{packet} \n\n")

pub.subscribe(onReceive, 'meshtastic.receive')
interface = meshtastic.serial_interface.SerialInterface()

time.sleep(1)
interface.close()
```

On a device that was just booted, I am indeed able to get stored messages. However, if I use the code referenced in the original issue, it instead does:

```
while True:
    time.sleep(1)
```

Never closes the connection. So what I think ends up happening is there is still a valid connection and and messages received are processed by the stale connection. So when I close that to try again, the message will never show up on the new interface.

I tried physically disconnecting the port and that doesn't seem to help. I can only seem to recover by actually restarting the meshtastic device.

Not sure if that is a python issue or firmware issue? or if there's a way to close the interface reliably

closed 2024-10-28 by kop316