meshtastic/python #688: Support tapback messages in mesh_interface.py

afourney, 2024-10-14:
I am building an application (APRS <-> Meshtastic gateway), that would significantly benefit from being able to send **tabpback messages**. For example, a tabpback message of thumbs up could signal that an APRS message was acknowledged on the other side of the gateway. For this I understand that I need to supply a replyId and emoji. 

Unfortunately mesh_interface.py only supports sending text messages. I would value either a tabpack method, or sample code for how I can create and send my own via data/protobuf.

ianmcorvidae, 2024-10-14:
The tapback messages are just text messages under the hood, but it does appear there's a couple properties it would be useful to be able to set without needing to use very low-level functions.

It appears that the iOS code (the only codebase that supports any sort of "tapbacks" -- neither android nor web has such a concept) sets the `emoji` property on https://buf.build/meshtastic/protobufs/docs/main:meshtastic#meshtastic.Data to `1` (which seems to confusingly be a bool very inefficiently stored in a 4-byte `fixed32`), and also sets the `reply_id` field of the same to a message that is being replied to. The combination of the two seems to be what iOS considers when deciding what to display as a tapback.

You could send this today with the python library by constructing a `Data` protobuf with those fields set, setting it as the `decoded` field of a `MeshPacket` (and setting other fields as needed, particularly the `payload` and `portnum` fields on `Data`, and the `channel` on `MeshPacket` if needed), then passing that to the `_sendPacket` function on the interface (along with other relevant arguments, if desired).

I'll look into adding helpers for these fields though, since ideally the underscore-prefixed methods wouldn't need to be used by end users.

afourney, 2024-10-14:
Thanks, I will give this a try!