meshtastic/python #693: [Bug]: meshtasic cli and web misses location set, import and export

xinrag, 2024-10-15:
### Category

Other

### Hardware

Linux Native

### Firmware Version

2.5.5.e182ae7

### Description

I guess LocalModuleConfig is outdated, does not know how to handle location.lot...:

Using a SX1262 hat for raspberry.

Found out that:
- web conf does not show alt, lon and lat if you try to set fixed position
- setting with android app ok, but then...
- meshtastic cli can successfull export this part in the yaml BUT...:
   location:
       alt: ...
       lat: ...
       lon:---
- meshtastic cli can not import this f.e. with --configure. Values get lost, not importet
- meshtastic cli can not this falues with f.e.:
    meshtastic --host 192.168.66.91 --set location.alt 401
  ..because "LocalConfig and LocalModuleConfig do not have attribute location.lat"

Atm don't know how to set position and othes without the android app.

And I would love it very much if --export-config can export all so that this is a backup of the config and can be used to restore, replace a hat

xinrag, 2024-10-15:
Phantastic! Now only the question with the web configuration... is there a way to get in in there, too?
I saw some older issues about that... looks like it is broken since some months.

ianmcorvidae, 2024-12-02:
The issue in the web interface would be https://github.com/meshtastic/web/issues/87 I think. I'll close this one, since the python CLI supports fixed positions just fine.

closed 2024-12-02 by ianmcorvidae