meshtastic/python #699: Segfault in active Serial Interface

kop316, 2024-10-22:
Hello!

When I have an active serial interface, after some amount of time, I get a segfault. When I look at the fault handler, I get this:

```
Fatal Python error: Segmentation fault

Thread 0x00007f3788e006c0 (most recent call first):
  File "/usr/lib/python3.12/threading.py", line 359 in wait
  File "/usr/lib/python3.12/threading.py", line 655 in wait
  File "/usr/lib/python3.12/threading.py", line 1431 in run
  File "/usr/lib/python3.12/threading.py", line 1075 in _bootstrap_inner
  File "/usr/lib/python3.12/threading.py", line 1032 in _bootstrap

Thread 0x00007f378b4006c0 (most recent call first):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 565 in read
  File "/usr/lib/python3/dist-packages/meshtastic/stream_interface.py", line 104 in _readBytes
  File "/usr/lib/python3/dist-packages/meshtastic/stream_interface.py", line 153 in __reader
  File "/usr/lib/python3.12/threading.py", line 1012 in run
  File "/usr/lib/python3.12/threading.py", line 1075 in _bootstrap_inner
  File "/usr/lib/python3.12/threading.py", line 1032 in _bootstrap

Thread 0x00007f37956006c0 (most recent call first):
  File "/usr/lib/python3.12/threading.py", line 355 in wait
  File "/usr/lib/python3.12/queue.py", line 171 in get
  File "/usr/lib/python3/dist-packages/meshtastic/util.py", line 287 in _run
  File "/usr/lib/python3.12/threading.py", line 1012 in run
  File "/usr/lib/python3.12/threading.py", line 1075 in _bootstrap_inner
  File "/usr/lib/python3.12/threading.py", line 1032 in _bootstrap

Current thread 0x00007f37b49b6080 (most recent call first):
  File "/usr/lib/python3/dist-packages/gi/overrides/Gio.py", line 42 in run
  File "/home/chris/dev/gtk-meshtastic-client/gtk_meshtastic_client/main.py", line 124 in main
  File "/home/chris/dev/gtk-meshtastic-client/./_build/gtk_meshtastic_client/local-meshtastic-client", line 58 in <module>

Extension modules: gi._gi, cairo._cairo, gi._gi_cairo, google.protobuf.pyext._message, _brotli, charset_normalizer.md (total: 6)
Segmentation fault
```

ianmcorvidae, 2024-10-23:
We don't have anything that's not pure python, so I don't think this can be an issue with this library directly. The current thread noted there is in some GTK code, which seems like the most likely issue to me -- pyserial is also, from what I can tell, not using any C extension modules. I guess there's also the possibility it's in the Google protobuf library that's noted in the list of extensions, but I'd definitely look at the GTK/GIO code first.

kop316, 2024-10-28:
Hello!

Sorry it took me a while to get back to you. Thank you for the info! I agree, I think there was a bug in GTK. I appreciate the feedback!

closed 2024-10-28 by kop316