meshtastic/python #703: Error handling makes it impossible to use as library

broglep, 2024-11-03:
The current approach of calling `sys.exit()` when a failure happens makes it impossible to use the library as part of an other project (e.g. home assistant integration)

Workaround of monkey patching that method does also not fully work as then the code flow continues and does not abort. 

Example: 
https://github.com/meshtastic/python/blob/master/meshtastic/mesh_interface.py#L444

ianmcorvidae, 2024-12-21:
Yes, definitely agreed. I'm curious what you think of the proposed refactoring in the comments on https://github.com/meshtastic/python/issues/579 -- I believe that'd also make it easier to update these cases. I haven't been able to devote much time to this library for the last few months, but hopefully I can devote some time in the next couple weeks.

broglep, 2024-12-21:
The general direction of #579 sound like a very needed improvement. There should be a clear split between mesh interface/connection/transport implementation and then the code that exposes this functionality as a cli (rendering)

Current situation of the library is pretty tangled, so much so that I didn't see a clear path on how to improve / work around it and I have made my own attempt at re-implementing parts of it: https://github.com/broglep/homeassistant-meshtastic/tree/dev/custom_components/meshtastic/aiomeshtastic

I also needed async support and focused on getting it working for long running connection, something that I didn't see in reach with the current code base. 

I found it actually quite hard to get a good abstractions and where to put what functionality, and the protobuf protocol itself is not helping, it is clearly visible that everything has grown organically. It would require quite some design work to build an interface/abstraction that abstract away all the messy detail of the protobuf protocol.