meshtastic/python #705: [Feature Request]: Get signal and environment telemetry

overlord23, 2024-11-22:
I'm looking to download environment metrics from a BME260 sensor and signal metrics like on the Android app.
The closest option I've found is `--request-telemetry` but includes only the device data

```Telemetry received:
Battery level: 91.00%
Voltage: 4.07 V
Total channel utilization: 0.00%
Transmit air utilization: 0.02%
Uptime: 466 s
```
It would be nice to have an option for both of them or integrate the data provided by `--request-telemetry`.

ianmcorvidae, 2024-11-22:
This is supported already; you can pass an argument to `--request-telemetry`, e.g. `--request-telemetry environmental`

closed 2024-11-22 by ianmcorvidae

overlord23, 2024-11-22:
I've checked the meshtastic-cli documentation but didn't find any relevant detail about the arguments available.
Thanks fo the hint