meshtastic/python #709: meshtastic.py tcp_interface.py maxing out CPU on one PC, unable to duplicate elsewhere

raudette, 2024-12-02:
I've written a script with meshtastic.py ( https://github.com/raudette/meshtastic_discord_bridge ):
- it works fine on my PC connecting with both USB and TCP, and have tested on a few others
- On one person's PC USB works fine, but TCP maxes out their CPU.  

Any guidance on how to resolve?
[image]

Details:

python -V
Python 3.11.2

cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"

Heltec Vision Master T190, firmware 2.5.2

ianmcorvidae, 2024-12-22:
I'm not really sure what to make of this; I certainly haven't seen it in other contexts either. I'm going to close this for now, but if you can find more information that might assist in figuring out the cause, or a reliable way to reproduce, feel free to leave a note here, reopen, or open a new issue with the added details.

closed 2024-12-22 by ianmcorvidae

derekn, 2025-04-08:
I am seeing this on 3 different RPi's when leaving the tcp connection open after about 10-15 minutes. im using latest meshtastic version and pre 2.6 and python 3.13

derekn, 2025-04-13:
following up on my last comment, this is reproducible by running `meshtastic --host <ip> --listen` and waiting a few minutes. my py-spy output is identical to OP...

```
  %Own   %Total  OwnTime  TotalTime  Function (filename)
100.00% 100.00%    9.98s     9.98s   _readBytes (meshtastic/tcp_interface.py)
 20.00%  20.00%    1.87s     1.87s   common (meshtastic/__main__.py)
  0.00% 100.00%   0.020s    10.00s   __reader (meshtastic/stream_interface.py)
  0.00% 100.00%   0.000s    10.00s   run (threading.py)
  0.00% 100.00%   0.000s    10.00s   _bootstrap (threading.py)
  0.00%  20.00%   0.000s     1.87s   main (meshtastic/__main__.py)
  0.00%  20.00%   0.000s     1.87s   <module> (meshtastic)
  0.00% 100.00%   0.000s    10.00s   _bootstrap_inner (threading.py)
```