meshtastic/python #712: [Bug]: Configuration doesn't fully restore

overlord23, 2024-12-17:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.5.11.8e2a3e5

### Description

The configuration doesn't fully restore on Heltec V3 and Heltec Wireless Stick Lite.
Here the applied steps to reproduce the problem

- issued nodedb reset and factory reset
- configured node and exported config with Meshtastic CLI
- issued nodedb reset and factory reset again
- exported config from Meshtastic CLI and compared with the previously exported yml

The following parameters included in the configuration file didn't restore:
- autoScreenCarouselSecs:
- txPower
- adcMultiplierOverride
- root (lora region)

Here the serial and debug log
[debug_aftrestore.txt](https://github.com/user-attachments/files/17910194/debug_aftrestore.txt)
[serial_log_cfgrestore.txt](https://github.com/user-attachments/files/17910195/serial_log_cfgrestore.txt)

This bug has been discussed in #5331 for reference.

ianmcorvidae, 2024-12-17:
@garthvh I think this is, in fact, a firmware bug. Python is just starting a settings transaction, sending the settings, and then committing; it seems like firmware is what isn't saving these values appropriately. There's been sporadic reports of this, particularly since 2.5.

@overlord23 if possible, could you also share the yaml file you're using to import, and the full output of `meshtastic --debug --configure <file.yaml>` (replacing `<file.yaml>` with your exported configuration, of course). I'd like to confirm that python is in fact sending the expected values, which don't all appear in the serial log.

lucaskatayama, 2024-12-21:
I get something like this...

Export config
1. Change mqtt.enable from true to false
2. Apply with --configure <YAML> --port /dev/<PORT>
3. Doesn't change anything...

BUT

Using bluetooth (--ble <name>), it works...
It only happens with serial port ....