meshtastic/python #728: [BUG] Config export encoding mismatch blocking import

todd2982, 2025-01-05:
When running --export-config the file generated is using UTF-16 LE encoding, but the --configure option used to import that same file only accepts UTF-8 encoded files.

My environment is python 3.13 running on Windows 11 (not within WSL) using Meshtastic module version 2.5.9.

My current workaround is to use VS Code to save the exported file using the correct UTF8 encoding.

Error during import:
```py -m meshtastic --ble myMACHere --configure .\SND5.yml
INFO file:ble_interface.py scan line:116 Scanning for BLE devices (takes 10 seconds)...
Connected to radio
Aborting due to: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
```

I'm unable to attach an export of the yml file due to github file restrictions.  I can send the file directly if requested.  I am also adding a link to a Stack Overflow thread that mentions the same issue with Python, but I'm not familiar enough with this module to make the changes myself.

https://stackoverflow.com/questions/3597480/how-to-make-python-3-print-utf8

todd2982, 2025-01-15:
Problem still exists in Meshtastic module version 2.5.10

pdxlocations, 2025-02-17:
I don't get an error as you described, rather the import appears to succeed printing all the values in the terminal.  Could your default encoding be set to UTF-16?

When I run
```
python3 -c "import locale; print(locale.getpreferredencoding())"
```
I get
```
UTF-8
```

And running:
```
file -mime example_config.yaml
```
gets:
```
example_config.yaml: ASCII text
```

I'm on a mac with python3.13.2, meshtastic-python2.5.11

todd2982, 2025-02-17:
You may have found my particular issue.  I do think a check in the Python CLI should be made when running the config export to check for this possibility, or hard-code the encoding in the export.  With the upcoming 2.6 release that will wipe all devices this is going to become a critical issue for those that may be in the same boat as me.

`py -V`
Python 3.13.0

`py -c "import locale; print(locale.getpreferredencoding())"`
cp1252

PR #798 (2025-07-22): Allow forced UTF-8 encoding for --export-config
This PR changes the --export-config command to save a UTF-8 encoded file instead of printing to stdout.  Allows the legacy use of > to redirect to a file.

To fix incorrect encoding on Windows Powershell:
```bash
meshtastic --export-config file.yaml
```
Also allow:
```bash
meshtastic --export-config > file.yaml
```

closes #728