meshtastic/python #729: serial port not found locally

DrBlackross, 2025-01-12:
install docker meshtastic on rpi, goto serial on the meshtastic client webpage, not found.... even after giving docker the serial port(s) where the lora is located locally in the system.

the docker image has no lsusb  or ability to update the docker image at all via remote root console, soo cant install lsusb or update.

kinda stuck trying to make it a stand alone in the corner mesh box, that i can add to the local net and check.

ianmcorvidae, 2025-01-12:
This doesn't appear to be a python issue either, since you're referring to the web client being run in docker.

The serial connection for the web client is a connection using web serial, which connects to serial devices on the local computer (the one loading the website), not on the server side where the client is being hosted.

To have what I think you're describing, a standalone box that's on the network, you need to either have the node itself directly on your network, ideally an esp32 with the web client included or a linux-native node, and use the HTTP endpoints with the web client, or you need to build something yourself. If you'd like to connect to a serial device connected to a network-accessible device, you'd need to build something in some way to forward that information from the serial device over the network, whether using socat or similar, or creating a proxy service that presents the serial client API as the HTTP client API.

closed 2025-01-12 by ianmcorvidae

DrBlackross, 2025-01-13:
my bad, i lost the post, lol.... ice in GA this weekend took out the internet :)