meshtastic/python #732: Timeout Error on rpi0 via serial

vogt31337, 2025-01-27:
Hi,

I am trying to use the api on my rpi0, but even listing the nodes crashes `(meshtastic --serial /dev/ttyUSB0 --nodes):`

```
Traceback (most recent call last):
  File "/home/dunai/venv/bin/meshtastic", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/dunai/venv/lib/python3.11/site-packages/meshtastic/__main__.py", line 1988, in main
    common()
  File "/home/dunai/venv/lib/python3.11/site-packages/meshtastic/__main__.py", line 1242, in common
    client = meshtastic.serial_interface.SerialInterface(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dunai/venv/lib/python3.11/site-packages/meshtastic/serial_interface.py", line 65, in __init__
    StreamInterface.__init__(
  File "/home/dunai/venv/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 55, in __init__
    self.connect()
  File "/home/dunai/venv/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 79, in connect
    self._waitConnected()
  File "/home/dunai/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1019, in _waitConnected
    raise MeshInterface.MeshInterfaceError(
meshtastic.mesh_interface.MeshInterface.MeshInterfaceError: Timed out waiting for connection completion
```
Running Rasbpian bookworm (Debian 12), installed meshtastic via apt-get
And meshtastic 2.5.18.89ebafc Beta on a Heltec Wireless Stick Lite V3
Using client.meshtastic.org it runs fine on my developer PC...

BR

putz612, 2025-02-09:
I am getting the same type of error but on Windows 11
```
PS C:\Users\user> meshtastic --version
2.5.11
PS C:\Users\user> meshtastic.exe --port COM5 --info
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\jsievert\AppData\Local\Programs\Python\Python312\Scripts\meshtastic.exe\__main__.py", line 7, in <module>
  File "C:\Users\jsievert\AppData\Local\Programs\Python\Python312\Lib\site-packages\meshtastic\__main__.py", line 1988, in main
    common()
  File "C:\Users\jsievert\AppData\Local\Programs\Python\Python312\Lib\site-packages\meshtastic\__main__.py", line 1242, in common
    client = meshtastic.serial_interface.SerialInterface(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\jsievert\AppData\Local\Programs\Python\Python312\Lib\site-packages\meshtastic\serial_interface.py", line 65, in __init__
    StreamInterface.__init__(
  File "C:\Users\jsievert\AppData\Local\Programs\Python\Python312\Lib\site-packages\meshtastic\stream_interface.py", line 55, in __init__
    self.connect()
  File "C:\Users\jsievert\AppData\Local\Programs\Python\Python312\Lib\site-packages\meshtastic\stream_interface.py", line 79, in connect
    self._waitConnected()
  File "C:\Users\jsievert\AppData\Local\Programs\Python\Python312\Lib\site-packages\meshtastic\mesh_interface.py", line 1020, in _waitConnected
    raise MeshInterface.MeshInterfaceError(
meshtastic.mesh_interface.MeshInterface.MeshInterfaceError: Timed out waiting for connection completion
```

erikwilson, 2025-03-06:
The issue for me (at least on MacOS  and soon to confirm on RPI) appears to be this block of code:
https://github.com/meshtastic/python/blob/ed908fc4b63b95d7948e969127331696df665610/meshtastic/serial_interface.py#L51-L57
After I commented out that code the meshtastic cli worked.

You will need to make sure your serial port is setup for BAUD_115200 correctly (for example on an e290):

`meshtastic --set serial.enabled true --set serial.rxd 44 --set serial.txd 43 --set serial.mode PROTO --set serial.echo 0 --set serial.baud BAUD_115200 --set serial.timeout 0`

(note that if you just use the number 115200 the serial module will fall back to 38400, it need to be `BAUD_115200`)

and then used a command like
`python ./meshtastic -s /dev/ttyS0 --info`

to successfully interact with the API (you will probably need to enable to serial module on the RPI also).

Liberty-Chris, 2025-04-11:
Similar scenario here. Odd thing is, I have 3 identical setups. Only one of them behaves this way. 

`admin@eventheim:~ $ meshtastic --port /dev/ttyUSB0 --set lora.tx_power 20
Traceback (most recent call last):
  File "/home/admin/.local/bin/meshtastic", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/admin/.local/lib/python3.11/site-packages/meshtastic/__main__.py", line 2062, in main
    common()
  File "/home/admin/.local/lib/python3.11/site-packages/meshtastic/__main__.py", line 1261, in common
    client = meshtastic.serial_interface.SerialInterface(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/.local/lib/python3.11/site-packages/meshtastic/serial_interface.py", line 74, in __init__
    StreamInterface.__init__(
  File "/home/admin/.local/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 55, in __init__
    self.connect()
  File "/home/admin/.local/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 79, in connect
    self._waitConnected()
  File "/home/admin/.local/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1073, in _waitConnected
    raise MeshInterface.MeshInterfaceError(
meshtastic.mesh_interface.MeshInterface.MeshInterfaceError: Timed out waiting for connection completion`

jacob-l-s, 2025-04-16:
I had the same problem,The device is RAK4631:
```
C:\Users\10842>ver
Microsoft Windows [版本 10.0.22631.5189]
C:\Users\10842>meshtastic --version
2.6.1
C:\Users\10842>meshtastic --port COM7 --info
Traceback (most recent call last):
  File "C:\Users\10842\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\10842\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\10842\AppData\Local\Programs\Python\Python310\Scripts\meshtastic.exe\__main__.py", line 7, in <module>
  File "C:\Users\10842\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\__main__.py", line 2062, in main
    common()
  File "C:\Users\10842\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\__main__.py", line 1261, in common
    client = meshtastic.serial_interface.SerialInterface(
  File "C:\Users\10842\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\serial_interface.py", line 74, in __init__
    StreamInterface.__init__(
  File "C:\Users\10842\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\stream_interface.py", line 55, in __init__
    self.connect()
  File "C:\Users\10842\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\stream_interface.py", line 79, in connect
    self._waitConnected()
  File "C:\Users\10842\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\mesh_interface.py", line 1073, in _waitConnected
    raise MeshInterface.MeshInterfaceError(
meshtastic.mesh_interface.MeshInterface.MeshInterfaceError: Timed out waiting for connection completion
```

Liberty-Chris, 2025-04-17:
The solution (and discovery) in my case was I could not set the region on a newly flashed device from CLI. And without that set, I couldn't do anything. I had to Bluetooth into the module and use the mobile app to set the region. I made my own firmware after to speed up subsequent device setups.

I could not do these from serial until manually enabled over Bluetooth:
meshtastic --port /dev/ttyUSB0 --set is_serial_enabled true
meshtastic --port /dev/ttyUSB0 --set lora.region 1