meshtastic/python #756: locationSource and LOC_EXTERNAL undocumented

roberthadow, 2025-04-04:
We need documentation of purpose and usage of locationSource and LOC_EXTERNAL.  

I have four Heltec V3s.  `meshtastic --info `shows that two have  `position.locationSource: "LOC_EXTERNAL"`  These two devices appear on maps.  The other two appear not to have latitudes and longitudes assigned, even though both are assigned in my CLI scripts.

I have no idea how I set or didn't set this attribute.

I tried to include ` --set position.locationSource 'LOC_EXTERNAL'` in my CLI script.  I got this error: `LocalConfig and LocalModuleConfig do not have attribute position.locationSource.`

garthvh, 2025-04-04:
This means it is a location provided by a phone gps generally, I don't think it is something you can affect from the API, it is in the protobufs and managed by the firmware.

ianmcorvidae, 2025-04-04:
Yes, `location_source` is a property on the position being sent out, not a configuration property. The apple and android apps set it on positions they acquire from the phone's GPS. If you're setting a fixed position (perhaps using `--setlat` `--setlon` and `--setalt` with the CLI), the correct location source for those is `LOC_MANUAL`, which the firmware sets on your behalf (the third option, `LOC_INTERNAL`, is what's used when a node has a physically attached GPS it runs itself). There's no other way to set or send a position with the regular CLI than the fixed position stuff. If you're not seeing those fixed positions appearing after you've set them, I recommend running `--remove-position` to unset the position and disable the fixed position setting (therefore enabling it to accept a new fixed position), then run a single command with all three of `--setlat`, `--setlon` and `--setalt` in one.

closed 2025-04-04 by ianmcorvidae

roberthadow, 2025-04-04:
Thank you, thank you , thank you @ianmcorvidae . Works like a charm.

If anyone else has a similar problem, let me relate my understanding of CRUD where the meshtastic CLI is concerned.

(1) It is efficient to build a list of the configuration attributes you want.  It is repeatable and can be reused to program multiple devices.  If you choose to produce the list programmatically, then execute it as a system command, that is an elegant approach. Example below.

(2) The list order is not important, but any attribute can only be mentioned once.  For example, you cannot remove an attibute then set it within a single list of arguments.  For example, `--remove-position` and `--setlon -99.99` cannot appear in the same argument list.  Nor can `--ch-index 0` and `--ch-index 1`.  You'll need to run meshastic twice to achieve your two different desires.

(3) The first time you run your script, check the results carefully.  Your fat fingers may throw an error.  Some incorrect commands are ignored silently.

(4) The meshtastic CLI does not follow quite the same rules as when you learned SQL. To wit:

  (C) Create.  You will likely never create a data element.  Meshtasic will have created it for you from its default selection.  Default values are often invisible and don't show up in the results of `meshtastic --info`.  They do exist, though.

  (R) Read.  Meshatastic has a robust getter.  At the terminal, enter `meshtastic` for a list of all the attributes.  Most every attribute can be read using this argument: `--get [config_section`]  or `--get [config_section].[option]`.    Latitude and longitude don't follow the pattern.  You may neither `--get position.longitude` nor `--get setlon`, but you can see them in the results of `meshtastic --info` if they are not blank.  One day, perhaps, all reads and writes wil follow the same pattern.

 (U) - Update.  Meshatastic has a robust setter.    Most every attribute can be set using this argument: `--set [config_section].[option]  [value]`.  Latitude, longitude, and altitude don't follow the pattern. they require setlat, setlong, and setalt. What is peculiar, though, is that the setter does not simply update an existing value.  Those must be removed first, using `--remove.position`. 

(D) Delete.  One doesn't so much delete a data element as return it to its default state.  This is the least well-developed process. For example, you cannot just `--set [config_section].[option]  DEFAULT` or `--set [config_section].[option]  ""`.  If you have a bunch to do, flash the device anew or use the factory reset option, then run your script with all of your preferences.  The one below works for me. I put the device-specific attributes at the top. The rest are the same for every device in my collection.

```
PARAMS=(
    --host 192.168.1.200
    --set-owner-short "BB38"
    --setlon -74.46477
    --setlat 40.89545 
    --setalt 154
    --set bluetooth.enabled true
    --set bluetooth.mode FIXED_PIN
    --set bluetooth.fixedPin 654321
    --set device.tzdef 'EST5EDT,M3.2.0,M11.1.0' 
    --set device.nodeInfoBroadcastSecs 90
    --set lora.region "US"
    --set module_config.ambient_lighting.green 70
    --set module_config.ambient_lighting.blue 70
    --set module_config.ambient_lighting.red 70 
    --set neighbor_info.enabled true
    --set neighbor_info.updateInterval 14400  
    --set network.addressMode STATIC
    --set network.wifiEnabled true
    --set network.wifiSsid 'Mine has spaces'
    --set network.wifiPsk mySecret
    --set network.ntpServer meshtastic.pool.ntp.org
    --set network.ethEnabled true
    --set network.ipv4Config.dns 16885952
    --set network.ipv4Config.gateway 16885952
    --set network.ipv4Config.subnet: 16777215
    --set position.position_broadcast_smart_enabled false
    --set position.position_flags 811
    --set position.positionBroadcastSecs 90
    --ch-set psk default --ch-index 0
    --reboot
)                                     
meshtastic "${PARAMS[@]}"

```