meshtastic/python #757: Windows - Cannot connect to seeed-xiao-s3 using serial port after upgrade from v2.6.0 to v2.6.1

krzotr, 2025-04-09:
I have upgraded the library to `2.6.1` using `pip install meshtastic==2.6.1` and cannot connect to my `seeed-xiao-s3` using the serial port. I have tested my other device - tbeam is working fine with `2.6.1`.

My platform:
- Windows 11 x64
- meshtastic --version: `2.6.1`
- device: seeed-xiao-s3
- firmwareVersion: `2.6.4.b89355f`

`test.py` code:

```
import logging

logging.basicConfig(
    level=logging.DEBUG,
    format="%(levelname)s file:%(filename)s %(funcName)s line:%(lineno)s %(message)s",
)

import meshtastic.serial_interface

iface = meshtastic.serial_interface.SerialInterface()
print(iface.getMyNodeInfo())
```

Running my code
```
pip install meshtastic==2.6.1
python .\test.py

DEBUG file:serial_interface.py __init__ line:36 ports:['COM11']
DEBUG file:serial_interface.py __init__ line:47 Connecting to COM11
DEBUG file:stream_interface.py __reader line:151 in __reader()
DEBUG file:stream_interface.py _sendToRadioImpl line:114 Sending: want_config_id: 2549725438
DEBUG file:stream_interface.py _sendToRadioImpl line:119 sending header:b'\x94\xc3\x00\x06' b:b'\x18\xfe\xf1\xe6\xbf\t'

Traceback (most recent call last):
  File "d:\dev\meshtastic\test.py", line 10, in <module>
    iface = meshtastic.serial_interface.SerialInterface()
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "d:\dev\meshtastic\.venv\Lib\site-packages\meshtastic\serial_interface.py", line 74, in __init__
    StreamInterface.__init__(
  File "d:\dev\meshtastic\.venv\Lib\site-packages\meshtastic\stream_interface.py", line 55, in __init__
    self.connect()
  File "d:\dev\meshtastic\.venv\Lib\site-packages\meshtastic\stream_interface.py", line 79, in connect
    self._waitConnected()
  File "d:\dev\meshtastic\.venv\Lib\site-packages\meshtastic\mesh_interface.py", line 1073, in _waitConnected
    raise MeshInterface.MeshInterfaceError(
meshtastic.mesh_interface.MeshInterface.MeshInterfaceError: Timed out waiting for connection completion
```

Also `--info` does not work

```
meshtastic --info
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "d:\dev\meshtastic\.venv\Scripts\meshtastic.exe\__main__.py", line 7, in <module>
  File "d:\dev\meshtastic\.venv\Lib\site-packages\meshtastic\__main__.py", line 2062, in main
    common()
  File "d:\dev\meshtastic\.venv\Lib\site-packages\meshtastic\__main__.py", line 1261, in common
    client = meshtastic.serial_interface.SerialInterface(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "d:\dev\meshtastic\.venv\Lib\site-packages\meshtastic\serial_interface.py", line 74, in __init__
    StreamInterface.__init__(
  File "d:\dev\meshtastic\.venv\Lib\site-packages\meshtastic\stream_interface.py", line 55, in __init__
    self.connect()
  File "d:\dev\meshtastic\.venv\Lib\site-packages\meshtastic\stream_interface.py", line 79, in connect
    self._waitConnected()
  File "d:\dev\meshtastic\.venv\Lib\site-packages\meshtastic\mesh_interface.py", line 1073, in _waitConnected
    raise MeshInterface.MeshInterfaceError(
meshtastic.mesh_interface.MeshInterface.MeshInterfaceError: Timed out waiting for connection completion
```

Reverting to 2.6.0 `pip install meshtastic==2.6.0` - everything is fine - can connect to `seeed-xiao-s3`:

```
python .\test.py
DEBUG file:serial_interface.py __init__ line:36 ports:['COM11']
DEBUG file:serial_interface.py __init__ line:47 Connecting to COM11
DEBUG file:stream_interface.py __reader line:151 in __reader()
DEBUG file:stream_interface.py _sendToRadioImpl line:114 Sending: want_config_id: 673038973
DEBUG file:stream_interface.py _sendToRadioImpl line:119 sending header:b'\x94\xc3\x00\x06' b:b'\x18\xfd\x84\xf7\xc0\x02'

DEBUG file:mesh_interface.py _handleFromRadio line:1258 in mesh_interface.py _handleFromRadio() fromRadioBytes: (...)
DEBUG file:mesh_interface.py _handleFromRadio line:1270 Received from radio: packet {
(.......)
```

CLI is working correctly

```
meshtastic --info
Connected to radio
(.......)

Owner: (...)
My info: (...)
Metadata: (...)
```

accik, 2025-04-11:
Had same issue with 2.6.1, downgrading to 2.6.0 worked instantly. I was connecting to a SenseCAP Card Tracker T1000-E using serial (com) on windows. 
```
 System: Windows
   Platform: Windows-11-10.0.22631-SP0
   Release: 11
   Machine: AMD64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v2.6.0 (*** newer version v2.6.1 available ***)
 Executable: C:\Users\asd\AppData\Local\Programs\Python\Python312\Scripts\meshtastic
 Python: 3.12.6 CPython MSC v.1940 64 bit (AMD64)
```

SAMS0N1TE, 2025-04-14:
Interesting, Tbeam supreme isn't working, but my heltec wireless stick v3 is fine.

ianmcorvidae, 2025-04-14:
yeah, I'm thinking that patch improved things for some nodes but broke it for the rest. I guess the simplest solution might be to hide that behavior behind a flag. There's another issue where folks were having issues even with the existing non-Windows equivalent, and I don't know if we can detect which setup is needed very easily.

todd-herbert, 2025-04-14:
I've always used that "wait to disconnect" argument to keep the port open when using devices which reset on DTR. It's not super elegant, but it does work.

PR #763 (2025-04-15): Revert "Work around RTS/DTR serial reset issues on Windows"
Reverts meshtastic/python#753
Fixes #757