meshtastic/python #759: Update Python CLI Guide

roberthadow, 2025-04-13:
The following meshtastic command arguments don't work reliably.  They either need to be fixed or the documentation updated:

`--test`  This argument throws the following error message: "Test module could not be important. Ensure you have the 'dotmap' module installed."

`--reply`   This argument results in a connection to the radio, then no further apparent action. Eventually, it fails with the following error message:
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/home/rhadow/miniconda3/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()

`--set-owner-short` This argument does not reliably reset the last four characters of the MAC address to uppercase. At least, it doesn't always stick.

`--dest ` user.id (i.e. `--dest '!b03bb4f8'`) works fine;  num (i.e. `--dest 2956702968`) doesn't.  The best you can hope for is "Received an implicit ACK. Packet will likely arrive, but cannot be guaranteed."  What is an _implicit ACK_?

rc14193, 2025-06-29:
I've started looking into this. The `--test` error is caused by dotmap not being installed because in the pyprojecttoml it is marked as optional. On a `poetry install` you need the `--all-extras` flag or removing the optional in the toml `dotmap = { version = "^1.3.30" }` Do you have a preference on which?

rc14193, 2025-07-04:
Ok I glazed over the make file. I see it uses pip by default. Maybe it's more that there needs to be a BUILD.md of some kind? The poetry lock file just stood out to me so I assumed poetry as the default and have been using that.

Is anyone really set on keeping dotmap? I think it could be eliminated since it's not pulled in the default pip install. There's other [onReceive functions](https://github.com/meshtastic/python/blob/master/meshtastic/__main__.py#L69)  that don't use it and from what I see test.py is the only place it's used (it's also in mesh_interface.py but those lines are commented out).

If Dotmap is still wanted the make file could be changes to add a new command for when someone wants to run `--test`

roberthadow, 2025-07-04:
As to _implicit ACK_, it seems that this is an acknowledgement that the message was received by some other node and perhaps retransmitted.  There's a big assumption that the destination node is alive in the mesh within _n_ hops. I suggest that _implicit ACK_ is an unfortunate choice of working. _Received and forwarded_  or _Receved and forwarded. Delivery not guaranteed_ would be better..