meshtastic/python #774: onReceive() question

mcc324, 2025-05-10:
Most of the time there is a 'decoded' item in the packet, but sometimes i see packets that don't contain that. What are those?

thanks,
Larry

pdxlocations, 2025-05-10:
They are likely packets sent with a channel psk that isn’t on your node and/or DMs not meant for you.

mcc324, 2025-05-10:
Here's the latest example. Just printing the info that might be relevant.

[2025-05-10 15:14:37] None --> ^all  snr = 6.0, rssi = -22   chan = 124

Due to the snr and rssi, I'm wondering if it might be some kind of packet from one of my nodes. There is no 'app' in the packet.

pdxlocations, 2025-05-10:
Portnums or Apps are in the encoded payload so you won't see them if your node can't decode it.

pdxlocations, 2025-05-10:
packets should either have:
```
encrypted: "\334\234\306\362\031I\230\024\217m\000\006\031I[\314\360r\024\rs32\273\326g}\351^!8\243\266 \033"
```
or something like:
```
decoded {
  portnum: POSITION_APP
  payload: "\r\000\240\025\033\025\000\340\332\266\030\273\001%!\266\037h(\003x\001\200\001\000\230\001\014\270\001\022"
  bitfield: 1
}
```
You'll only see decoded if your node has the key