meshtastic/python #775: readGPIOs function

jlrasmussen, 2025-05-12:
Not sure if this should be an issue. I'm trying to get a value back from the readGPIOs(id, mask) function in the remote_hardware.py module. I'm not sure if I'm using it correctly. 

iface = MagicMock(autospec=SerialInterface)
rhw = RemoteHardwareClient(iface)
# rhc = remote_hardware.RemoteHardwareClient(interface)
destination_node_id = 1128161828
bitmask = 16
value = 1
rhw.watchGPIOs(destination_node_id, bitmask)
rhw.writeGPIOs(destination_node_id, bitmask, value)
rhw.readGPIOs(destination_node_id,bitmask,True)

How do I get the gpio_value from the read function? 

Or should I be doing this some other way?

garthvh, 2025-05-13:
Remote hardware is disabled by default, did you build a custom firmware with it enabled?

jlrasmussen, 2025-05-13:
I also put this on github

Also, didn't mention it but the writeGPIOs(dest, mask, value) doesn't set
the masked pin high either.
Yes, did the following based on meshtastic
For firmware versions > 2.5.3, this module requires you to compile your own
firmware and remove the -DMESHTASTIC_EXCLUDE_REMOTEHARDWARE=1 build flag in
platformio.ini. Additionally, while configuring this module may be
available in clients, setting and reading GPIO's is currently only possible
using the Meshtastic Python CLI
<https://meshtastic.org/docs/software/python/cli/>

Also, use meshtastic --set remote_hardware.enabled true which enables the
remote hardware verified with --info.
"remoteHardware": {
"enabled": true,
"allowUndefinedPinAccess": true,
"availablePins": [
{
"gpioPin": 4,
"name": "relay ",
"type": "DIGITAL_WRITE"
}
]
},

Pin 4 has been set to OUTPUT
Using meshtastic --gpio-wrb 4 1 --dest !433e6224 turns a relay on but the
hdw.writeGPIOs(dest,mask, 1) doesn't
meshtastic --gpio-wrb 4 1 --dest !433e6224
Connected to radio
Writing GPIO mask 0x10 with value 0x10 to !433e6224

Similarly meshtastic --gpio-rd --!433e6224 reads the value
meshtastic --gpio-rd 0x10 --dest !433e6224
Connected to radio
Reading GPIO mask 0x10 from !433e6224
Received RemoteHardware type=READ_GPIOS_REPLY, gpio_value=16 value=16

However, the meshtastic CLI commands don't give me direct access to the
value read. I would like to be able to use the direct python functions
readGPIOs to get a value back that I can use.

jlrasmussen, 2025-06-02:
Another week. Any update on this Garth? Please reply that it's not going to be fixed or
if it's not considered a problem.