meshtastic/python #778: How to setup the ham parametes?

mcc324, 2025-05-18:
I haven't been able to find a way to turn off 'ham mode' from the cli. If i do get it set. how do you read it back to verify?

Larry

pdxlocations, 2025-05-18:
Ham mode sets your longname to your callsign and turns off encryption.  To undo this you just have to set an encryption key on your channels and optionally rename your node.

```
To disable Ham mode, use --set-owner and --set-owner-short to reset the owner names, and use --seturl or --ch-set commands to configure channels with the name and encryption you wish them to have. Or, use --factory-reset to reset to default settings.
```
From:
https://meshtastic.org/docs/software/python/cli/#--set-ham-set_ham

SubhaNAG2001, 2025-07-12:
To manage and verify the 'ham mode' setting in Meshtastic via the CLI, follow these steps:

### 1. **Turn Off Ham Mode**
   Use the `--set` flag to disable ham mode:
   ```bash
   meshtastic --set ham false
   ```
   - Replace `/dev/ttyUSB0` with your device port if needed (e.g., `--port /dev/ttyUSB0`).
   - For remote devices, use `--host <IP/HOSTNAME>`.

### 2. **Verify Ham Mode Status**
   Use the `--get` flag to check the current setting:
   ```bash
   meshtastic --get ham
   ```
   **Output Example**:
   ```
   ham: false
   ```

### 3. **View All Settings (Optional)**
   To see all device configurations (including `ham`):
   ```bash
   meshtastic --getall
   ```

### 4. **Check CLI Version**
   Ensure your CLI is up-to-date (as shown in your command):
   ```bash
   meshtastic --version
   ```

### Key Notes:
- **Permissions**: Use `sudo` if you encounter permission errors (e.g., `sudo meshtastic --set ham false`).
- **Reboot**: Some devices reboot automatically after changes; if not, power-cycle manually.
Device Connection**: If settings aren’t applied, confirm your device is connected properly (check port/host).

garthvh, 2025-07-14:
> To manage and verify the 'ham mode' setting in Meshtastic via the CLI, follow these steps:
> 
> ### 1. **Turn Off Ham Mode**
> 
> Use the `--set` flag to disable ham mode:
> 
> meshtastic --set ham false
>  * Replace `/dev/ttyUSB0` with your device port if needed (e.g., `--port /dev/ttyUSB0`).
> * For remote devices, use `--host <IP/HOSTNAME>`.
> 
> ### 2. **Verify Ham Mode Status**
> 
> Use the `--get` flag to check the current setting:
> 
> meshtastic --get ham
> **Output Example**:
> 
> ```
> ham: false
> ```
> 
> ### 3. **View All Settings (Optional)**
> 
> To see all device configurations (including `ham`):
> 
> meshtastic --getall
> ### 4. **Check CLI Version**
> 
> Ensure your CLI is up-to-date (as shown in your command):
> 
> meshtastic --version
> ### Key Notes:
> 
> * **Permissions**: Use `sudo` if you encounter permission errors (e.g., `sudo meshtastic --set ham false`).
> * **Reboot**: Some devices reboot automatically after changes; if not, power-cycle manually.
>   Device Connection**: If settings aren’t applied, confirm your device is connected properly (check port/host).

This is just for individual settings and does not use the ham parameters packet