meshtastic/python #784: [Bug] waitForTraceRotue reset does not do what is intended

grleblanc, 2025-06-09:
We have code that performs traceroutes to various nodes (with respect to mesh load, etc). After running our service for any period of time, we see the below error.

Exception: 

```
  File "/usr/local/lib/python3.13/site-packages/meshtastic/mesh_interface.py", line 652, in sendTraceRoute
    self.waitForTraceRoute(waitFactor)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/meshtastic/mesh_interface.py", line 1011, in waitForTraceRoute
    success = self._timeout.waitForTraceRoute(waitFactor, self._acknowledgment)
  File "/usr/local/lib/python3.13/site-packages/meshtastic/util.py", line 229, in waitForTraceRoute
    self.reset()
    ~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/meshtastic/util.py", line 203, in reset
    self.expireTime = time.time() + self.expireTimeout
                      ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
OverflowError: int too large to convert to float
```

Upon printing out the `self.expireTimeout` value, I can see it continuously increase by a factor of 10 every time we perform a traceroute.

Looking at the code, the `reset()` method only resets the `self.expireTime` and not the `self.expireTimeout` that is being increased by a factor of `waitFactor`

```
    def reset(self) -> None:
        """Restart the waitForSet timer"""
        self.expireTime = time.time() + self.expireTimeout
```

```
    def waitForTraceRoute(self, waitFactor, acknowledgment, attr="receivedTraceRoute") -> bool:
        """Block until traceroute response is received. Returns True if traceroute response has been received."""
        self.expireTimeout *= waitFactor
```

If I manually edit the file and set `self.expireTimeout` to `20` it fixes the problem.

I wanted to make a issue first to confirm this is an issue and discuss posible solutions before making a PR.

Cheers!

ianmcorvidae, 2025-06-09:
I think you're right -- we're modifying the timeout value but not resetting it after.

I'd love to redo a lot of this timeout logic, but for just fixing this for now, I wonder if the easiest option would be to add an optional argument to `reset` that defaults to the `expireTimeout` and then use that in `waitForTraceRoute`, something like:

```python
    def reset(self, expireTimeout=None):
        self.expireTime = time.time() + (self.expireTimeout if expireTimeout is None else expireTimeout)

# ...

    def waitForTraceRoute(self, waitFactor, acknowledgment, attr="receivedTraceRoute") -> bool:
        """Block until traceroute response is received. Returns True if traceroute response has been received."""
        self.reset(self.expireTimout * waitFactor)
        # ...
```

What do you think of this?

grleblanc, 2025-06-09:
I think that LGTM.

I just updated my local install here and it's working perfectly.  Let me make a PR.