meshtastic/python #788: Man Page

jp-bennett, 2025-06-12:
Is it practical to create a man page for the Meshtastic command? Would be a nice addition if there's a standard way to do it for Python packages.

Aareon, 2025-06-19:
PyPI: `argparse-manpage`

This can be handled via GitHub Actions and triggered automatically.

roberthadow, 2025-07-02:
I use `meshtastic --help` as the current and authoritative command reference.  

In order for that to work, there needs to be an online source to tell us what is the most current version of meshasticd. Or in the alternative, an apt-update (or equivalent) that guarantees you get the most current version.

roberthadow, 2025-07-05:
@jp-bennett I suggest that the matter of Meshtastic documentation be loooked at holistically.  Why do people look at documentation? Because they are trying to implement some feature for themselves or troubleshooting a problem.

Documentation includes every bit of human digestable information.  It has a range from close to the code to distant, including --help and error messages, to Meshtastic.org pages, to sources over which the developers have no control, including Reddit and Stackoverflow threads, to random YouTube videos.

It's really important to attach human-readable messages to a date or version.  Nothing worse that Googleing a problem ande coming up with a thread five years old.  The challenge is increased by Google AI that summarizes incorrectly already bad information. 

The suggestion of a man page is great.  We need to make sure when it is implemented that --help, the Meshtastic.org web pages, and the man page are all sourced from the same place and identified by version and device. Else, ther errors and omissions will propogate more than they have today.