meshtastic/python #808: [Bug]: Error while parsing FromRadio bytes

bplein, 2025-07-28:
### Category

Other

### Hardware

Other

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.8, 2.7.0, 2.7.1

### Description

While this is an error in the Python CLI, it seems to also affect iOS clients as well. Not reproducible with Web client.

Also reproduced on RAK, Lilygo T-Beam Supreme, and Linux Native nodes.

A while back, I ran into this, and didn't open an issue at that time. I am hitting it again with newer firmware and CLI and client versions.

There is a node in our mesh that seems to be using some odd character code in its short name that causes errors with the CLI. This node does NOT show up in iOS node listing, even though it's in the node and causing these issues via CLI.

The CLI example of the error follows:
```
meshtastic -t p3n0 --get 0
ERROR file:mesh_interface.py _handleFromRadio line:1264 Error while parsing FromRadio bytes:b'"o\x08\x94\x9d\xec\xcd\t\x12D\n\t!99bb0e94\x12\x05TTT01\x1a\x04T01\xe0"\x06\x80e\x99\xbb\x0e\x94(2B +\t\x0e\x8be\x817\xa1F\x97*v\xed\x89\xa0A\xdf4\'\xc5\xabP\xb9TF\xd0\xed9Q`e\x1c%\x00\x00\x82\xc1-\x9b\x046h2\x15\x08e\x15\x1f\x85\x9f@\x1d\x00\x00\x00\x00%i\xb7\xa2?(\xac\xb6\tH\x02' Error parsing message with type 'meshtastic.protobuf.FromRadio'
Traceback (most recent call last):
  File "/Users/bill/miniconda3/lib/python3.9/site-packages/meshtastic/mesh_interface.py", line 1262, in _handleFromRadio
    fromRadio.ParseFromString(fromRadioBytes)
```

This error will hit fairly consisently when running a variety of `meshatastic` CLI commands. It happens over TCP (as shown above), BLE or Serial.  Only by using `--no-nodes` could I be successful running CLI commands.

In the case above, I could go to the webUI and see the node, but couldn't get it from the command line.

[image]

I had no way of reaching the user (they are not on our community discord, and I wasn't close enough to message them via Meshtastic), so I deleted the node entry and blocked their incoming packets across all my nodes and went on with life.

Recently I removed them from the block, and the bug is hitting me again.

Let me know if there''s more I can share. Currently blocking them again for the time being.

### Relevant log output

```Shell

```

ianmcorvidae, 2025-07-28:
It's unfortunately pretty hard to fix this. Their shortname, expressed in python bytes as `b"T01\xe0"`, isn't valid utf-8 since `e0` can't appear in that encoding, but there's no way to override the interpretation of strings as utf-8 in the python protobuf library, and it doesn't seem to support sending us the actual unicode decoding error, just that DecodeError.

If we can figure out what process is allowing invalid utf-8 to be in shortnames that would maybe be good, possibly in firmware somewhere. Unsure.

bplein, 2025-09-01:
> It's unfortunately pretty hard to fix this. Their shortname, expressed in python bytes as `b"T01\xe0"`, isn't valid utf-8 since `e0` can't appear in that encoding, but there's no way to override the [...]

Should this be fixed in the firmware instead? Some client in use is trying to shove in an illegal name (or a one time bug allowed it to be sent). The firmware should filter that out (or is the firmware promiscuous in that it trusts all clients to do the right thing, and doesn't sanitize inputs?)

cpatulea, 2026-01-15:
```
$ meshtastic --version
2.7.6
$ meshtastic  --serial --info --no-nodes
Connected to radio

Owner: Meshtastic <snip> (<snip>)
My info: { "myNodeNum": 1136054892, "rebootCount": 552, "minAppVersion": 30200, "deviceId": "G+uKcNpxHdT+KNz3OqwjYg==", "pioEnv": "heltec-v3", "nodedbCount": 1, "firmwareEdition": "VANILLA" }
Metadata: { "firmwareVersion": "2.7.15.567b8ea", "deviceStateVersion": 24, "canShutdown": true, "hasWifi": true, "hasBluetooth": true, "positionFlags": 811, "hwModel": "HELTEC_V3", "hasPKC": true, "excludedModules": 1280, "hasEthernet": false, "role": "CLIENT", "hasRemoteHardware": false }

Nodes in mesh: {
  "!<snip>": {
    "num": 1136054892,
    "user": {
      "id": "!<snip>",
      "longName": "Meshtastic <snip>",
      "shortName": "<snip>",
      "macaddr": "<snip>",
      "hwModel": "HELTEC_V3",
      "publicKey": "dhXcfvkdu+LUZCGYVzjTiqRhNLOcsq6U+26rJhbe700=",
      "isUnmessagable": false
    },
    "position": {},
    "deviceMetrics": {
      "batteryLevel": 101,
      "voltage": 4.49,
      "channelUtilization": 3.3633332,
      "airUtilTx": 0.028027777,
      "uptimeSeconds": 61
    },
    "isFavorite": true
  }
}

Preferences: {
  "version": 0
}

Module preferences: {
  "version": 0
}

Channels:
  Index 0: PRIMARY psk=default { "psk": "AQ==", "moduleSettings": { "positionPrecision": 13, "isMuted": false }, "channelNum": 0, "name": "", "id": 0, "uplinkEnabled": false, "downlinkEnabled": false }
ERROR file:mesh_interface.py _handleFromRadio line:1302 Error while parsing FromRadio bytes:b'2@\n-Fr' Error parsing message with type 'meshtastic.protobuf.FromRadio'
Traceback (most recent call last):
  File "/Users/<snip>/.local/pipx/venvs/meshtastic/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1300, in _handleFromRadio
    fromRadio.ParseFromString(fromRadioBytes)
google.protobuf.message.DecodeError: Error parsing message with type 'meshtastic.protobuf.FromRadio'
ERROR file:stream_interface.py __reader line:204 Error while handling message from radio Error parsing message with type 'meshtastic.protobuf.FromRadio'
Traceback (most recent call last):
  File "/Users/<snip>/.local/pipx/venvs/meshtastic/lib/python3.12/site-packages/meshtastic/stream_interface.py", line 202, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/Users/<snip>/.local/pipx/venvs/meshtastic/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1306, in _handleFromRadio
    raise ex
  File "/Users/<snip>/.local/pipx/venvs/meshtastic/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1300, in _handleFromRadio
    fromRadio.ParseFromString(fromRadioBytes)
google.protobuf.message.DecodeError: Error parsing message with type 'meshtastic.protobuf.FromRadio'

Primary channel URL: https://meshtastic.org/e/#CgcSAQE6AggNEgA
```

Looks like some kind of incomplete read:

```
DEBUG file:mesh_interface.py _handleFromRadio line:1296 in mesh_interface.py _handleFromRadio() fromRadioBytes: b'2E\n-FromRadio=STATE_SEND_QUEUE_STATUS, numbytes=6\x15,phi\x1a\rSerialConsole \n'
DEBUG file:mesh_interface.py _handleFromRadio line:1308 Received from radio: log_record {
  message: "FromRadio=STATE_SEND_QUEUE_STATUS, numbytes=6"
  time: 1768452140
  source: "SerialConsole"
  level: DEBUG
}

DEBUG file:mesh_interface.py _handleFromRadio line:1296 in mesh_interface.py _handleFromRadio() fromRadioBytes: b'2E\n-Fr'
ERROR file:mesh_interface.py _handleFromRadio line:1302 Error while parsing FromRadio bytes:b'2E\n-Fr' Error parsing message with type 'meshtastic.protobuf.FromRadio'
```

bplein, 2026-05-17:
Closing. Don't care.

closed 2026-05-17 by bplein