meshtastic/python #819: Lack of documentation/Help for settings options

Marx1, 2025-08-30:
Related to https://github.com/meshtastic/meshtastic/issues/2005

There is a lack of built-in documentation for what settings accept what values. for example if you list the settings with --set you just get a list of settable settings, but not what acceptable inputs are:

[image]

Once you read or set something you only get the raw value ie:

[image]

And the --info returns in a JSON with no acceptable input values.

I'm recommending two changes:

Change 1:

Add an option to display the acceptable input for a given command. taking the bluetooth settings from https://github.com/meshtastic/meshtastic/issues/2005:

```
bluetooth:
    bluetooth.enabled
             0 = Bluetooth Disabled
             1 = Bluetooth Enabled
    bluetooth.fixed_pin
            xxxxxx = 6 digits for pin
    bluetooth.mode
           0 = Random Pin
           1 = Fixed pin (Requires fixed_pin to be set)
           2 = No Pin
```

Change 2: 

Modify how the --set and --get command respond to include a human readable format as well. Something like this:

[image]

SpudGunMan, 2025-08-30:
This might help? 

https://meshtastic.org/docs/configuration/radio/bluetooth/

roberthadow, 2025-11-04:
@SpudGunMan Hello Kelly --

The point here is that the --help, the web-based documentation (https://meshtastic.org/docs/configuration/radio/bluetooth/), and the CLI error messages should all be generated from a common repository, so that one is not forced to look in different places..  The format of all the commands should be the same, i.e. --set preference PREFERENCE.  The names of the preferences should all be in a common format, e.g. camelCase or snake_case, but not a mixture.  PREFERENCES can be alpha or numeric, but human readable alpha is preferable. For every set method, there should be a corresponding get method.